subutai-io/blockchain-router #2: Schematic Design of Subutai Router

bkukanov, 2017-10-09:
The SR 2.0 adds extra add-on features to SR 1.0, as well as it enhances the Ethernet Switch capability of SR 1.0 from Fast Ethernet to Gigabit Ethernet.
The board adds a Raspberry extension connector connected to its Ultrascale FPGA PL I/Os, it also has an Arduino Uno R3 compatible extension connector which is also connected to PL FPGA I/Os. These 2 connectors, together with the 5 PMOD extension connectors gives user the ability to plug n play the COTS products to his SR 2.0.
- Our peripherals needed to check compatibility

metinburak, 2017-10-17:
@bkukanov  let's get this waterfall. The blok diagram will be in the PCB folder. Create the folder and push the block diagram pdf into there.

metinburak, 2017-10-17:
[sR2BlockDiagram.pdf](https://github.com/subutai-io/subutai-router/files/1392735/sR2BlockDiagram.pdf)

metinburak, 2017-10-17:
I have added @bkukanov .

closed 2017-10-17 by metinburak