subutai-io/blockchain-router #13: TPM interface schematic capture

metinburak, 2017-10-17:
The TPM interface as header will be added to the design with this issue.

bkukanov, 2017-10-30:
Here little research about TPM header
https://superuser.com/questions/1216494/what-is-the-difference-between-various-tpm-modules
- ASUS TPM-M R2.0 (14 pin)  ->  TPM 2.0 compatible
- Asus TPM-L R2.0 (20 pin) ->  TPM 2.0 compatible
- Asus Trusted Platform Module TPM/FW3.19 (20 pins) -> TPM 1.2 compatible
- Gigabyte TPM Module GC-TPM2.0 (20 pins)

Asus header pinout http://forums.overclockers.com.au/threads/trusted-platform-module-asus-z170m-e-d3.1182000/

bkukanov, 2017-11-02:
I think it's better to use the header for 20 pin, because there is some compatibility with Asus TPM & Gigabit's TPM

bkukanov, 2017-11-02:
https://www.gigabyte.com/Motherboard/GC-TPM-rev-20#ov
http://download.gigabyte.ru/manual/mb_manual_ga-b250-hd3p_e.pdf
[image]

bkukanov, 2017-11-06:
done beside the eMMC header sheet

closed 2017-11-06 by bkukanov