subutai-io/blockchain-router #22: Generating the DDR4 pinout in Vivado's latest (for Ultrascale)

metinburak, 2017-10-17:
The #9 needs this in order to connect the DDR4 module sch to Zynq. We will create a new project with Vivado based on Ultrascale and get the pinout.

metinburak, 2017-11-22:
@bkukanov we need this to route the PL SODIMM and also, any PL ios assigned so far in the pcb will be affected by this default pins generated by the PL ddr controller in vivado ultrazynq edition. What do u think?

bkukanov, 2017-11-23:
ok @metinburak  will look into it

bkukanov, 2017-11-27:
Here is the pin out of 64bit
[example_design_64bit.txt](https://github.com/subutai-io/subutai-router/files/1504894/example_design_64bit.txt)
[image]

bkukanov, 2017-11-27:
here is pin out for 72bit
[image]
[example_design_72bit.txt](https://github.com/subutai-io/subutai-router/files/1504899/example_design_72bit.txt)

closed 2017-11-27 by bkukanov